sta-titansrobotics/19446-IntoTheDeep #2: Intake

mrfabroa, 2024-09-12:


GianBangayan, 2024-09-12:
Intake ideas 
- bare minimum extension half a tile
- max extension should only be one tile max

3 main grabbing ideas
- claw 
- side rollers (bennefit of centering)
- top down roller

Random idea
- make it offset for cycling from bucket due to how field is structured, less movement would be needed

GianBangayan, 2024-09-12:
[image]

GianBangayan, 2024-09-14:
[https://drive.google.com/file/d/16pY2o-AI32QQMUqpbcSkL976ApIZk1p5/view?usp=drive_link](url)
[https://drive.google.com/file/d/15CgS3h90axSpMLBP9xsHopG8bGpqgMJW/view?usp=drive_link](url)
[https://drive.google.com/file/d/1xfal_sp3z6guyiXjoOe6IDXQhvIbpKD1/view?usp=drive_link](url)
Ethan F. + Gian - Top-down roller intake testing w/ funnel for centering

GianBangayan, 2024-09-14:
[https://drive.google.com/file/d/16pY2o-AI32QQMUqpbcSkL976ApIZk1p5/view?usp=drive_link](url)
Ethan S. + Cameron + Erick - Side roller intake, two sets for centering

Ethan-Selvarajah, 2024-09-14:
https://drive.google.com/file/d/16NDgVqlQM1P7j6Wb3oTZvZbjZaoVHEFq/view?usp=sharing
**Side Roller (Car Wash) Intake:**
- Two sets of rollers, one for initial intake, and then one to reorient it.
- Main inconsistencies were a result of the nature of the prototype (cardboard... eg. in the video, we noticed better results after taping the cardboard). Keep in mind that there is still _some_ inconsistency due to the rollers not being locked in place.
- In future iterations, it is likely that we would decrease the size of the first set of rollers.
- This design may not be the most optimal because having multiple sets of rollers increases points of failure and more parts are needed for implementation.
- When observing this intake's most optimal runs, it is extremely fast.

(Ethan S. + Cameron + Erick + Mariya [with some discussion with Gian later])

MariyaVla, 2024-09-14:
Link to summarized intake ideas: https://docs.google.com/document/d/1AwJEi2_UoQM0G6PpzjqCMNelnguDJZk6yCiNc0I_mA0/edit?usp=sharing

Ethan-Selvarajah, 2024-09-14:
**Passive Claw Intake:**
https://drive.google.com/file/d/1ZJFUr8hI4QSVKqmm2m3pkx-jAn8XPDdV/view?usp=sharing

This is the first prototype of the passive claw idea, CAD by @GianBangayan.

**What Works:**
- The initial grabbing of the sample works well. 

**What Needs Improvement:**
- When placing the sample on the bar, the claw needs more force on the sample to ensure that it takes more force from the bar pushing on the sample to release it. This flaw is demonstrated in the video.
- The living hinges on the claw may wear down over time. This can even be seen through the "wiggle test" in the video, and was noticed even more later on.

There is already another iteration in progress.

mrfabroa, 2024-09-15:
For the side roller solution, one of the key benefits is that upon intake the sample is aligned.   My question is, is alignment a key requirement?? If so, why?

Ethan-Selvarajah, 2024-09-15:
My perspective is that although the orientation of the samples is still important later on (when placing the samples into the bucket because we want to maximize space), it's not needed during intake. I think it's actually better to do this somewhere in the transfer. For example, in both mine and @GianBangayan's ideas (refer to #6), the orientation is fixed in the transfer. If we do this then we won't need the second set of rollers on the intake, which means fewer parts that can malfunction/break, less complexity, and less weight. Also, I feel like since the intake has to extend out, it's going to be more vulnerable compared to the rest of the robot. That just makes it more fragile, so we definitely want simplicity regarding the intake.

Ethan-Selvarajah, 2024-09-18:
[image]
@GianBangayan intake sketch

mrfabroa, 2024-09-21:
@Ethan-Selvarajah thoughts on how to align the sample post-intake/pre-intake?

closed 2024-10-01 by mrfabroa