sta-titansrobotics/19446-IntoTheDeep #4: Climb

Erick-Meza, 2024-09-13:


Erick-Meza, 2024-09-14:
The current idea for climbing is to utilize a double hook mechanism to achieve 3rd level ascent. Hooks are currently designed to be 3D printed, but using metallic hooks could result in less stress due to material strength if the opportunity to use metal arises. If planning on attempting 3rd level ascent, it would be beneficial to modify the wall mounts on the robot to have an angle that reduces friction and makes climbing easier (see FRC video below).

[image]

For 2nd level ascent, a singular hook mechanism will suffice to get the robot to hang on the metal bars. There are currently 2 potential variations within ideation: 

[image]
The first variation is to use one large hook powered by 2 viper sliders to achieve hang. Using 1 hook with 2 sliders will make the mechanism powerful and capable of lifting the robot with less stress on individual parts.

[image]
Another variation is to use 2 hooks, each mounted on a slider. While this does put more individual stress on thinner hooks, this can be solved by using stronger material or a higher infill %. In addition, a hang design using 2 hooks has proven to be successful when attempting to hang.

These current ideas have been inspired by FRC Team 254: https://youtu.be/eYPNndagILM?si=A87qzHTZsUAG3pNM

GianBangayan, 2024-09-15:
[image]
Lower set of hooks can be set on a hinge system, so when the bar gets pushed on top of it, it moves out of the way and laches on after unfurling.
Moves out of the way to let the bar and once the bar clears it enough it will fold back out and hook onto the bar.

or have like a thing that when the bar presses a button the hooks flip out and get locked in or smth.
[image]

Ethan-Selvarajah, 2024-09-18:
We will be using 435 RPM motors since it is the closest to 1 m/s speed. This is optimal for the distance that we wish to travel, and should have enough torque to support our robot's weight during the climb.

Refer to #1 for calculation.

Erick-Meza, 2024-09-20:
Hooks can be created differently to assist the climb for each ascent level. The top set of hooks can be designed like single or eye hooks to make grabbing the pipes easier:

[image] 
[image]

Since the lower set of hooks will need to support the robot's weight, it may be beneficial to go for a choker hook design for better security/strength:

[image]

In addition, it may be beneficial to design the hook with an angle that will allow the robot to move upwards without the hooks impeding the motion smoothly.

Hook designs were researched here. The designs were made for cranes but can still be effective within our robot: https://www.maximcrane.com/blog/crane-rigging-hooks-types-and-applications/

mrfabroa, 2024-09-21:
> it moves out of the way and laches on after unfurling.

@Erick-Meza look into the use of springs for this.  Maybe even some rotational springs

closed 2024-10-01 by mrfabroa