sta-titansrobotics/19446-IntoTheDeep #5: Outtake

GianBangayan, 2024-09-14:


GianBangayan, 2024-09-14:
Initial claw revision (V1)
[image]
- Clips onto specimen from wall
- Potential for breakage along clip wall
- Lack of auto-centering/funnel in front

Claw V2
[image]
- improves life of repeated use 
- Reduces force for pulling down specimens
- added funnel auto-centering

Claw V3
[image]
- Improves down pull force
- changes? (not sure if it's better or worse) lifespan

GianBangayan, 2024-09-14:
Claw V3 (old) Intial Tesiting 
https://github.com/user-attachments/assets/c8df09ea-2396-4eb5-b9fa-038dab20cf89
- noticed it didn't have enough down-force to pull it down
- Noticed wear on joints/lines indicating wear after 50+ uses

GianBangayan, 2024-09-14:
Bucket is very reliant on the intake method and transfer, so further information on that is needed before proceeding with bucket design.

3 Options
- Container + arm 
Pros
- simple
- arm helps reach upwards towards bucket and increases speed
- fast/reliable
Cons
- questionable rule-breaking? (letting it go/ it flying/falling?)
- programmers have to make macro/harder for driver to aim
- questionable transfer

- Container + flaps/tubing + arm
[image]
Pros
- Same as above
- Holds blocks for a few ms longer before releasing
Cons
- same (without Rule breaking)
- May need more force from the arm to release

- Container + active release
Pros
- simple
- fast/reliable
- easy to aim/control/less risk of drop/consistency
Cons
- questionable transfer
- slower than one with an arm

mrfabroa, 2024-09-21:
@GianBangayan need to document the use of the claw for outtake.  As in Ethan F's transfer claw

GianBangayan, 2024-09-24:
We decided to go with an active claw after discussing potential robot designs to aid in a faster and more consistent transfer.

GianBangayan, 2024-09-24:
Version 1 Claw, used 1 or 2 servos? was confused about power transfer from one arm to another, Didn't want to use gears.
[image]

Version 2 Claw, uses 1 servo, using a crankshaft to power both arms, entirely 3d printed, lack of mounting points
[image]

Version 3 Claw, better version 2 claw, servo is mounted to a metal bracket to the rest of the plastic structure, it's stronger overall and the plastic is lighter. Has a metal mounting point and can be connected to arm easily.
[image]

closed 2024-10-01 by mrfabroa