sta-titansrobotics/19446-IntoTheDeep #6: System Integration

GianBangayan, 2024-09-14:


GianBangayan, 2024-09-14:
[image]
General robot rough drawing from Gian.
- Top-down tubing intake
- slide intake extension
- Vertical sliders 
- two stage hang, (slides + winch)
- outtake arm
- Passive specimen placer + bucket for samples

GianBangayan, 2024-09-14:
[image]
[image]
General robot Rough drawing from Ethan S.

- Ethan pls fill this out
- Telescopic intake extension
- Claw with rotating wrist mechanisim

GianBangayan, 2024-09-14:
[image]
General robot rough drawing from Ethan F.

Ethan-Selvarajah, 2024-09-15:
**Initial Thoughts:**
My initial idea was the use of an arm that functioned similarly to Team [16379 Kookybotz's robot](https://www.youtube.com/watch?v=mT0KNbaz_80) from Centerstage last year. Specifically, I thought the use of a telescopic arm + claw would be interesting/optimal.

**Here are a few sketches of those initial ideas:**
[image]
[image]

**Based on these sketches, I had this idea for sample collection:**
1. The arm extends and the claw grabs the sample.
2. The claw has a servo which rotates 90*, and simultaneously the arm will retract into the robot.
3. The sample is then funneled into the robot where the transfer takes it in.
4. It is taken in using a compliant wheel and/or rotating surgical tubing.
5. The funnel ends up at the bottom of the sliders, which hosts a deposit box.
6. The sliders extend upwards and then rotates the box to slide it into the buckets.

I discovered this page for a [telescopic arm](https://www.thethriftybot.com/products/thrifty-telescoping-tube-kit). I watched their assembly video and found it both interesting and useful (if this idea was to be pursued).

**Hang Process:**
1. A hook attached to sliders.
2. [This](https://www.youtube.com/watch?v=IgF5ukoe12Y&feature=youtu.be) also attached to sliders somehow? Or maybe somewhere else. Need some way to ensure that the robot stays straight, otherwise the hang will be misaligned. 

The hang process is not refined either, and requires more thought.

**Additional Notes:**
These were just some of my initial thoughts, but it may not be the most efficient. I noticed that telescopic arms seemed very fast and stable, and the idea stemmed from there. Also, I am unsure if the transfer process would actually be efficient.

--Accidentally posted this under #1 but I'll move it here instead.

closed 2024-10-01 by mrfabroa