sta-titansrobotics/19446-IntoTheDeep #10: Sprint 1 - Review and Demo

mrfabroa, 2024-09-23:


MariyaVla, 2024-10-04:
For the Chamber, there were a lot of successes and it helped tie the whole build together. The dimensions worked well with the claw from the outtake as well as the Ramp coming in. For the future it would be good to add some tolerance and space for the sample so that it could be easier to transfer from the ramp into the enclosed box. Since the trapdoor is going to be edited in the next iterations, it will still undergo a couple changes, but the overall build of the idea has been working well.

MariyaVla, 2024-10-04:
For the overall intake there were many challenges but also a lot of accomplishments! We started off with an intake that would flip both the chamber and the ramp to go over the barrier and we had made a functional design that connected the pick up, transfer, trapdoor, and outtake; most components had been printed out and tested in person with some success! but after some consideration of the design, near the end of the due date, we ended up deciding on only turning the ramp since it would help with mounting motors and being able to manage the weight being flipped and manipulated. Because of this change in idea, we unfortunately have to change a many components, which means that the intake had not been finalized in the past 2 weeks. But we mean to implement the new changes into CAD and printing them out to find more areas of improvement in the next couple of weeks.

Ethan-Selvarajah, 2024-10-04:
**Trapdoor:**
**Successes:**
- The mechanism worked smoothly
- Sample stopper on the inside helped keep it centered despite it's longer length
- No interference with claw coming up

**Failures:**
- The intake is changing (the chamber is no longer rotating), which has reduced the need for a trapdoor
- The way the rubber band tensioner was mounted was not ideal
- - Fragile
- - Placement prevented optimal direction of force (too close to point of rotation)

**Next Steps:**
- Redesign the trapdoor
- - Potentially just have one smaller flap (the one closer to the entry point of the sample) and remove the second flap

Honestly, in terms of next steps, just removing one trapdoor technically works, and it can be optimized from there. However, as the chamber gets updated, the trapdoor may be further updated as well. There have been discussions of widening the chamber, which would be easy to update. Additionally, the trapdoor has to be created in a way so it doesn't block the sample from actually entering.

cameronso5, 2024-10-04:
pick up
Successes
- made a mounting for the servos 
- made only the ramp move up and down and not the chambermade 
- a new design that fixed a lot of the mounting issues we had
- found a way to mount the servos to the sliders 

Failures
- had to redo intake because of some issues with flipping the channel and ramp. 
- The ramp was too thin and too short. 
- had  issues with mounting the motor with the old design 

Next Steps:
- The next step is to find out how to move the spiny things with a motor. 
- Make the ramp longer so it would not vary steep when going over the barrier.
- Finish the mounting with the motor.

Erick-Meza, 2024-10-07:
###Hang/Hook Design

Success:
- Created a functional design for robot to safely hang from
- Designed a simple yet effective way of mounting the custom hook to the sliders
- Found and developed a way to ensure the robot does not fall as a result of swinging/pendulum motion that could result in the hooks sliding off the pole.

Failures:
- Did not consider the mounting/placement of the hooks relative to the entirety of the robot (how it would like relative to drivetrain, 3 stage sliders, etc.)
- While the design is secure and reliable, it is not the most efficient for the drivers in comparison to other designs (such as the ladder design that was only slightly explored/developed), as it requires more movement and precision from the drivers.
- Since the design uses rubber bands for the one way trapdoor, it is possible they snap/break and require fixing, which can be tedious and unnecessary during competition.

Next Steps:
- Create the sprint 1 design into real life and limit test for possible flaws/improvements (Strength of trapdoor, strength of overall hook design, limit of weight the hook can support, etc.)
- Look into ladder design as a possible upgrade (3D print it and compare to original design to determine which is more effective as both a method of hanging and as an easy task for drivers to complete)

GianBangayan, 2024-10-09:
**Claw/arm**
 4th claw CAD design worked after printing and was proven to be able to accomplish all tasks.
Pros
- was able to complete all objectives

Cons
- integrity and repeated use in question
- partial ply within parts (fixed with tweaking

closed 2024-10-16 by mrfabroa