sta-titansrobotics/19446-IntoTheDeep #13: Pick Up

MariyaVla, 2024-09-24:


cameronso5, 2024-10-01:
I started to work on the intake park of the robot by designing the sides that will funnel the specimen into the chamber. 

[image]
V1

cameronso5, 2024-10-01:
I made some changes to the intake by changing the floor to make it into a ramp and also added a weel hole so that it can support the ramp on the grond and to make it easier for the drivers to move the intake when it is in the submersible. I also added holes for the hex shafs and fixed some issues with the intake. 

[image]
v2

cameronso5, 2024-10-01:
add a hex shaft and the weels 
[image]
to the intake to make sure it fit

cameronso5, 2024-10-01:
I worked on the motor placement for the rolers. I placed it there because it needs to move with the sliders and the intake needs to flip. I added some more support so it will help it for moving a lot when it is on. 

[image]

closed 2024-10-16 by mrfabroa