sta-titansrobotics/19446-IntoTheDeep #15: Trapdoor

MariyaVla, 2024-09-24:


Ethan-Selvarajah, 2024-09-24:
[image]
[image]
V1 Trapdoor

Ethan-Selvarajah, 2024-09-25:
[image]
[image]
- Made it more compact
- Added space in the bottom so it can move over top of the corner and not get caught
- Moved positioning of the shaft higher so that it doesn't interfere with sample entering (one more thing I want to discuss tomorrow)

Ethan-Selvarajah, 2024-09-25:
[image]
- Added hooks for rubber banding
- Reduced thickness of trapdoor

Ethan-Selvarajah, 2024-09-26:
https://github.com/user-attachments/assets/10935598-aa44-4242-a4d2-7be8d72ef35a
Things to Fix:
- Turn the hex holes in the corners into 4mm circles so screws can go in
- Potentially move the hooks and holes (for the rubber band) to the center of the side of the trapdoor so that it's able to pull it back fully
- If I do the step above, then consider shortening the base of the rotation points on the corner
- Make the chamber longer so that the sample contacts a point closer to the center so it faces less resistance
- - Think: Pushing something closer to its rotation point vs. pushing something farther from its rotation point. 

Also, note for the video, that isn't how we will actually be rubber-banding it, it won't go all the way around.

Ethan-Selvarajah, 2024-10-01:
[image]
[image]
- Trapdoor was made longer to fulfill the longer chamber

Ethan-Selvarajah, 2024-10-02:
Due to the intake design change regarding the rotation, the use of a trapdoor is now minimal. The only real usage is to prevent the sample from flying out, however, this is unlikely to occur. It seems as though the negatives (inefficiency) outweigh the benefits. If not removed, at the very least, the trapdoor will have to be updated, perhaps as some sort of flap near the intake (as if it does get flung, this is where it's most likely to happen).

Ethan-Selvarajah, 2024-10-04:
[image]
One change was also made to the way the rubber bands work because that method was prior to snapping.

However, trapdoors are still being significantly revamped.

closed 2024-10-16 by mrfabroa