sta-titansrobotics/19446-IntoTheDeep #19: Durable Climbing Design (Climb/Hang) - Sprint 1

Erick-Meza, 2024-09-24:
Durable Climbing Design (Climb/Hang) - Sprint 1

_Originally posted by @Erick-Meza in https://github.com/sta-titansrobotics/19446-IntoTheDeep/issues/9#issuecomment-2371938994_

Erick-Meza, 2024-09-24:
For this sprint, the goal of the climb will be to have a working prototype completed on Onshape. This includes the overall shape of the hook and mounting design for the sliders. In addition, the goal consists of having a version 3D printed and tested for strength limits/design performance.

Erick-Meza, 2024-09-26:
Day 1:

- Created 2 different iterations/versions of the claw to test curvature/flow of the design
- Began designing a way to create a trapdoor to improve strength when climbing and eliminating the possibility of the robot falling off
- The trapdoor will fit perfectly into the design of the overall claw to reduce space taken from trapdoor when open. In addition, the door will use torsion springs or a rubber band system to allow  the bar to enter the hook.
[image]
[image]

mrfabroa, 2024-09-28:
@Erick-Meza where does it mount on the robot and relative to the bar how does it line up when the robot is parked and ready to climb?   Do you have a sketch for that?

Erick-Meza, 2024-10-04:
Hook will be mounted near the bottom of the 1st stage of the slider mechanism using a flat beam. This increase proximity with the robot and bring the center of gravity closer towards the hooks. The idea behind this design is to raise the sliders to pipe level, then proceeding to ram towards the pipe. The trapdoor opens inwards and secures the pipe within the system, to where the robot will then hang.
[image]

Erick-Meza, 2024-10-04:
During the final days of the sprint, a ladder design mentioned on discord was also investigated and prototyped. The overall design/lifting mechanism would provide the drivers with an easier hang experience, since the hinge system will reduce the need for precision. A concern Ethan F had with the design is that it wouldn't replicate the "locking/flush" seen in the ladder design because of the circular application within FTC.

See ladder mechanism inspiration here: https://www.youtube.com/watch?v=ZTnK2fytD3s&t=2m09s
[image]

closed 2024-10-16 by mrfabroa