sta-titansrobotics/19446-IntoTheDeep #22: Sprint 2 - Review and Demo

mrfabroa, 2024-10-07:


Ethan-Selvarajah, 2024-10-19:
**Intake:**
**Successes:**
- Major aspects of intake are done
-- Ramp is done
-- Mounting is fine
Connection to chamber is done

**Failures:**
- Sample currently gets stuck in funnel
- Rollers too high

**Next Steps:**
- Fix angle of funnel and offset the ends of the funnel to allow the sample to enter without getting stuck
- Change motor mounting

-@Ethan-Selvarajah @cameronso5

Erick-Meza, 2024-10-22:
##Hook Design

###Success:

- Hook was able to support the relative weight of the robot
- Design can also account for possible swinging of the robot

##Failures

- Hook requires some precision from the drivers in order to use

##Next Steps

- [ ] Redesign the hook in a way that does not require precision from drivers (completed during long weekend with Ethan Fung's approval on new design)
- [ ] Mount hooks onto robot for completion of Hang and test hook design using a completed robot

MariyaVla, 2024-10-29:
**### CHAMBER**
The chamber has went through multiple iterations and the final version has been designed and is ready to print

### Successes
- Chamber Holds Pixel Well
- Overall Shape of trapdoor is achieved
- Final shape of chamber is achieved
- Design of elastics going in and through the chamber has been implemented
- The trapdoors have one range of motion to open and can do so easily in theory

### Failures
- The design has not been printed, so there is possible issues that will only arise after printing

### Next Steps
- Print out the Chamber and trapdoors
- Check the rubber band placement and if it works
- Look for any fixes needed

closed 2024-11-04 by Ethan-Selvarajah