sta-titansrobotics/19446-IntoTheDeep #24: Intake

MariyaVla, 2024-10-16:


MariyaVla, 2024-10-16:
This week we have made a new design to better implement a ramp and add all the new components and remove the previous ones that were making the intaking process more difficult. On our new intake design, we are implementing a detached ramp which is going to be the only movable part. This will help the ramp go over the barrier surrounding the samples without having to use a motor to move the whole component including the actual motors and surgical tubing mounting. The Ramp is attached on a singular servo since it is light weight. Then, there is a component that is used as walls on the side of the ramp which are crooked and different angles so that the sample wouldn't get stuck sideways during intake. These angles allow the sample to rearrange itself to be carefully placed into the chamber with the top wall holding down the sample so that it doesn't slip out through the top. After this intake the sample ig going to go into the chamber which is attached to the sliders with standoffs. 

[image]

Ethan-Selvarajah, 2024-10-22:
[image]
[image]
- Funnel was updated
- - Different angles on either side to help guide the sample inwards
- - Roller mount was lowered so that it can reach the floor and past the ramp
- - Roof was removed
- Ramp was updated
- - Dimensions changed to make it fit more cohesively with the funnel

-@cameronso5 @Ethan-Selvarajah

MariyaVla, 2024-10-23:
For chamber I have made unjustements to add some trapdoors to help hold the sample down rather than fly out during the intake. I also changed the adjustments to fit one claw prong rather than two we had before. So in summary:

- Shortened Chamber
- Added Trapdoors for Transfers
- Change from 1 cut out to 2 cut outs for new claw
- Adjust mounting of screws with new trapdoor

[image]

MariyaVla, 2024-10-28:
Keeping in mind the objectives of the chamber and trap door:
- The sample should have no restrains in passing into the chamber
- The sample shouldn't be able to escape the chamber until the claw pulls it out
- The claw should be able to easily grab the sample without getting caught or stuck
- The trapdoors should easily open and snap back into place at a 90 degree angle, without going 100% back down

The previous chamber has went through printing and has been edited to fix more of its flaws throughout the past few days. 
- I made sure that the fit of the trapdoor wouldn't be too tight onto the chamber 
- I flipped the position of the hook so that it would be easier to create tension and easier to print 
- I added two holes so that the rubber band could pass through the trapdoor and out onto the outside of the chamber (keeping in mind that I don't want the rubber bands to stop the sample from passing in)
- I adjusted the side of the ingrave for the claw since it was too small and had gotten stuck a couple times when testing grabbing mechanism
- I removed an area of the fillet from the inside of the trapdoor so that instead of going all the way back down and hitting the wall, it would get stopped by a flat surface and would stop at a 90 degree angle
- Finally, I added a small notch at the back so that i could tie off the rubberband there, this will be printed to test out and go through iterations

[image]

[image]

[image]

closed 2024-11-04 by Ethan-Selvarajah