sta-titansrobotics/19446-IntoTheDeep #29: Sprint 3 - Review and Demo

mrfabroa, 2024-10-28:


Ethan-Selvarajah, 2024-11-01:
**Intake:**
_Note: This does not include information on the chamber & trapdoor._

**Successes:**
- Intake is essentially completed
- - The ramp is completed and printed.
- - The funnel is completed and printed.
- - Intake runs smoothly.
- - - Everything works efficiently and as expected -- refer to "#30" description for what that means (I will expand on a few issues in the next two sections).

**Improvements & Next Steps:**
- Samples can get stuck on the sides of the intake, as seen in testing. However, this is not a big deal. The only thing that needs to be changed is the number of surgical tubing pieces we have. Initially, we were planning on utilizing 3 surgical tubing pieces in the middle areas, but now we are going to have 5 (just two more added on either side) to force the sample through.
- There was some worry with structural integrity due to some flex scene with the 3D printed parts. However, this was improved with the addition of metal (quad blocks) and also with the beams added later on (both on top and below the intake).
- Mount the intake fully onto the robot (not just on sliders for testing, like it is currently) and drive around and test it.
- We have the wrong size optake for our intake (it is too large). This is very simple and should be completed very quickly. The only reason it hasn't been done already is because our 3D printing people have been busy.
- Chain our rollers together.

-@Ethan-Selvarajah @cameronso5

MariyaVla, 2024-11-02:
### **CHAMBER & TRAPDOORS / TRANSFER**
Refer to #30 for details!

The basic goal of these components from the start was to allow the sample to be quickly placed in a container of sorts, from there moved around without a risk of it falling out, and finally a way for the sample to pass out of the container with ease. For this we were set on a chamber design with trapdoors. Throughout the sprints I had taken over this role and had created the finalized version of the component. 

### SUCCESSES
- The chamber efficiently picks up the sample without it getting caught (Before testing with intake)
- The chamber can move around side to side and up and down without the sample falling out (only through the intake hole which isnt an issue)
- The chamber aligns with the mounting of the sliders efficiently
- The chamber also aligns with the cut outs that are needed for the claw and the outtake
- The trapdoors have a good resistance that helps keep sample stable in movement 
- The trapdoors snap back into place at a perfect 90 degree angle without going inwards
- The trapdoors are 100% passive which is a benefit
- The rubber bands are well tensioned and are quick and easy to replace between games ( if they end up breaking, its much easier to repair than an active mechanism)

Overall there have been very many successes with the designs and all the objectives I have previously listen in the sprints have been achieved! 

### FAILURES AND NEXT STEPS
When it comes to failures, there are none with the current design. We will be testing the transfer with intake and outtake soon and maybe then I will have to do some minor edits. It is possible that when adding the intaking components such as high speed surgical tubing and an angled ramp, the transfer might not be smooth and it may have to be fixed. Or I may have to lower the resistance so that the outtake can actually pick up the sample with the high tension. But overall, there seems to be no flaws currently.

- @MariyaVla

Erick-Meza, 2024-11-04:
## Hang

### Successes

- Created an efficient and simple to utilize design for drive team
- Printed design in a way that is space efficient while retaining the same strength as previous iterations
[image] [image]

### Failures

- Lowest screw hole was created in context to the incorrect screw, resulting in a hole that was too small for the screw to fit in. While the hook can still be mounted, it is not placed as far down on the vertical sliders as originally planned.

### Improvements/Next Steps

- Fully mount both sliders in their correct positions on the drivetrain
- Keep the hook mounted on the sliders for programmer reference
- 3D print new hooks with correct hole diameter

closed 2024-11-05 by mrfabroa