sta-titansrobotics/19446-IntoTheDeep #30: Sprint #3 - intake

cameronso5, 2024-10-28:


MariyaVla, 2024-10-29:
https://github.com/user-attachments/assets/aba9acc6-f395-4367-a38c-4ca10610388a

Ethan-Selvarajah, 2024-10-31:
https://github.com/user-attachments/assets/eabcc2b5-b846-4087-afd8-3b0749a7e203

We tested our intake in the field.

We noticed that we need to:
- Add two more surgical tubing pieces on the front roller (on the left and right) so it prevents the samples from getting jammed
- Add servo or some sort of clamping mechanism to chamber so that we can easily outtake any other unwanted samples
- Make the intake stronger because right now when its jostled it feels very flimsy (more metal)
- Trapdoor needs to be updated (completed now)

-@Ethan-Selvarajah @cameronso5

Ethan-Selvarajah, 2024-10-31:
We are also waiting for an updated optake (smaller) to be printed because the current optake is too large, preventing proper testing.

-@Ethan-Selvarajah @cameronso5

closed 2024-11-05 by mrfabroa