sta-titansrobotics/19446-IntoTheDeep #43: Intake Revision (Sprint 5)

Erick-Meza, 2024-11-29:


Ethan-Selvarajah, 2024-12-12:
**Chamber Revision Decisions:**
Chamber has undergone a wide array of changes since the last sprint. These decisions were influenced by the experience noted at the competition, and by some ideas we had pre-Kingston that couldn't be implemented in time. These changes involved the shortening of the chamber, reduction of height, removal of trapdoors, update of mounting points, and adjustment for the new claw. In this update, I will go into depth on each of these revisions.

**Shortening of the Chamber:**
The chamber's length was shortened for a multitude of reasons, but mainly to allow the sample to be reached by the intake. Due to the nature of our intake, being active, rather than something more precise such as a claw, there is a chance of the incorrect colour being taken in. (Of course, the intake allows for near-instant intake, so it's still beneficial, and can be resolved with driver practice). By allowing it to hang out into the intake, it can be spun out if the intake spins in reverse. It is essential to note that this will still work because the tubing from the intake will lock the sample in place based on its rigidity.

**Reduction of Height:**
The chamber's height was reduced mainly because there was no real need for the walls to be particularly high. 

**Removal of Trapdoors:**
This change was expected. The trapdoors were only really needed in our initial idea, where the entire intake rotated 180*. The trapdoors were only kept because we thought it could aid in holding the sample in, but after analyzing it during our matches, it did not truly have any impact. As such, we decided to remove it. It added in unneeded complexity.

**Update of Mounting Points:**
The entire metal foundation of the intake changed. It was a significant source of our weight. Due to this, the mounting points also changed.

**New Claw Slit Size Adjustment:**
With @GianBangayan's new claw, the size of the chamber also had to be changed. In my initial updates, I wanted to make the chamber smaller, but with the 5 cm width of the claw, the slit had to be larger.

_(will add process pictures)_

Ethan-Selvarajah, 2024-12-12:
Original Chamber:
[image]

New Chamber:
[image]
[image]

Notice how the sample sticks out slightly, the different mounting points, the removal of the trapdoors, the smaller height, and the different slit size.

Ethan-Selvarajah, 2024-12-12:
**Chamber with Claw Assembly:** 
[image]
Note: I did consider leaving one side of the wall since the claw only needs one side now, but, in the end I decided to keep it. The reason I chose to do this was because the other wall isn't needed anyways, and the small portion on the bottom is enough to keep it in.

Ethan-Selvarajah, 2024-12-12:
**Chamber Future Steps:**
- Consider lowering the back wall
- - That full height may not necessarily be required either, but further testing is needed (maybe the force of the intake will send it flying slightly upwards and the wall height could help catch it?)
- Test it with the rest of the intake as a whole

Ethan-Selvarajah, 2024-12-12:
**Chamber Pictures:**
[image]
[image]
[image]
[image]

Ethan-Selvarajah, 2024-12-12:
### Ramp Revision Decisions:
Ramp underwent some changes in this sprint. This time it was primarily based on the experience we had at the Kingston competition.

Physical Hardstop:
A physical hardstop was implemented to ensure that after a certain amount of rotation, the ramp wouldn't be able to move any further. This would allow for accuracy in rotation that we couldn't achieve with programming. Additionally, this helped mesh better with the chaining approach we had with intake due to inaccurate movemens we had there.

I believe some minor changes to the dimension occurred as well.

Ethan-Selvarajah, 2024-12-12:
### Intake Foundation Decisions:
The intake underwent significant changes in its foundation. These changes stemmed from our experiences at Kingston. A major issue that our robot faces is its extreme weight. The weight prevents us from hanging (@Erick-Meza has some torque calculations that can be referenced, in which it was determined that we must lose 8 kg for the motors to have enough torque). We noticed that the intake had a lot of unnecessary metal foundation (excessive amounts of quad blocks...).

**Weight Reduction through Quad Block Removal:**
@cameronso5 and I brainstormed alternative methods to remove weight from the intake. We decided to use an L flat pattern bracket. This is significantly thinner then the quad blocks we utilized. We were initially going to keep 1-2 quad blocks for the bearing, but decided to remove that as well because we realized we could add the bearing directly into the L flat pattern bracket.

**Slider Changes:**
The slider stages was updated from 3 to 2. This is because we noticed that when the intake extended, the weight of the intake on the end of the sliders caused it to sag. Additionally, 3-stage sliders were not required. The sliders were also rebuild and restrung. This is because at the competition there were multiple issues with the sliders, specifically with incorrect screw lengths and even a random screw being stuck between the sliders. By completely rebuilding it, we were able to control the quality (this was done amongst multiple people as well).

Ethan-Selvarajah, 2024-12-12:
**Ramp Images:**
[image]
[image]

**Ramp Images Assembly:**
[image]

Ethan-Selvarajah, 2024-12-12:
### Tubing Info:
At the moment, I'm looking for a more rigid tubing. I've found [this](https://www.mcmaster.com/51225K25/), but I'm looking for another supplier for a better price. Our current tubing has a durometer of 40A, while this is 65A.

Old Tubing:
McMaster Carr, durometer is 40A, standard for intake Super-Soft Latex Rubber Tubing for Air&Water, 3/16" ID, 3/8" OD ([OLD](https://www.mcmaster.com/5234K25))